vial-kb/vial-gui #322: Key Override does not work with Delete key as replacement

el-frantastico, 2025-04-27:
I'm attempting to have Shift + Backspace be Delete. Thought I'd use the Key Override feature, however I can't get it to work.  I know that the Key Override feature in general works because if I set the Replacement as another key, say the character A, it correctly does Backspace or A depending on whether Shift is not held or held, respectively.

**Repro Steps**
1. Open _Key Override_ tab.
2. Enable an unused key override page.
3. Set _Backspace_ as the _Trigger_.
4. Check _LShift_ and _RShift_ for _Trigger mods_.
5. Set _Delete_ as the _Replacement_.
6. Open text editor and type a random line.

**Expected Behavior**
When _Backspace_ is pressed while _Shift_ is not held, _Backspace_ is sent and the previous character is removed. When _Backspace_ is pressed while _Shift_ is held, _Delete_ is sent and the next character is removed.

**Actual Behavior**
When _Backspace_ is pressed while _Shift_ is not held, _Backspace_ is sent and the previous character is removed. When _Backspace_ is pressed while _Shift_ is held, nothing happens.

xyzz, 2025-04-27:
include a screenshot of https://config.qmk.fm/#/test or similar key logger when pressing the override

Masca1919, 2025-11-03:
I had this same issue but found out I had not toggled "suppressed mods" to *L/RShift* so the output key was actually still registering the modifier (Shift+Delete).
Therefore, it worked inconsistently, sometimes not doing anything at all (MS Word) which gave me the impression it was not working at all. Maybe that's it?

[image]