vial-kb/vial-gui #325: Add quantum buttons for Gui+Alt?

jakepearson, 2025-05-16:
I have a few commands that use cmd+alt+<something>. I can define them in vial with an arbitrary value, but then I need if I look at the layout I see hex codes like 

[image]

Could there be quantum buttons for `GA` like there are for most of the rest of the modifier combos? Not sure how easy that is, but if you would take a PR, I could give it a shot if you point me toward an example.

skoch13, 2025-08-21:
There are plenty of shortcuts that use gut + alt/opt on macOS, so quantum for this will be beneficial

arturoszulc, 2025-09-14:
Would like that quantum too, since Im using tab switching in VSCode and it is gui + alt + arrow

mikejhill, 2026-01-05:
I am similarly curious about quantum buttons for `RCtrl + RAlt + <Key>`. Is there a QMK limitation around this, or is it just something that would have to be added to Vial?